robotpy/mostrobotpy #15: Pin robotpy-build version

TheTripleV, 2023-11-02:


virtuald, 2023-11-13:
The individual packages already do this.

closed 2023-11-13 by virtuald

TheTripleV, 2023-11-13:
The problem is that the setup.py in each package import robotpy_build, which it can't do if it's not installed. I had to manually install robotpy_build after cloning this repo when I first tried this. So rdev should be smart enough to install the correct version of robotpy_build.

virtuald, 2023-11-13:
There is already a [install_build_deps](https://github.com/robotpy/mostrobotpy/blob/main/devtools/subproject.py#L41) function in Subproject that can be used, but only CI currently does that. I don't know whether it would make more sense to add it as part of the develop command or as a separate build-deps command.