robotpy/mostrobotpy #33: RobotPy startup on a RoboRIO is slow

virtuald, 2023-11-13:
On my RoboRIO with a 2016 image, I get the following (mostly consistent timing between runs, no DS connected):

```
roboRIO-2423-FRC:~$ time python3 -c 'import hal'

real	0m4.954s
user	0m1.280s
sys	0m0.170s
roboRIO-2423-FRC:~$ time python3 -c 'import wpilib'

real	0m6.984s
user	0m3.270s
sys	0m0.260s
```

HAL library loading may be a big chunk of this:

```
roboRIO-2423-FRC:~$ time python3 -c 'import ctypes; ctypes.CDLL("/usr/local/lib/python3.5/site-packages/hal_impl/libHALAthena.so")'

real	0m3.093s
user	0m0.470s
sys	0m0.070s
```

As a baseline:

```
roboRIO-2423-FRC:~$ time python3 -c 'pass'

real	0m0.210s
user	0m0.180s
sys	0m0.020s
roboRIO-2423-FRC:~$ time python3 -c 'import pip'

real	0m3.002s
user	0m2.760s
sys	0m0.220s
```

It would be good to reduce the startup time somehow.

auscompgeek, 2024-01-02:
> It has occurred to me that a large chunk of the import time is probably pybind11 creating docstrings. Would probably be good if that could get deferred until they're actually accessed.

It turns out that nanobind defers docstring rendering, where pybind11 doesn't: https://nanobind.readthedocs.io/en/latest/why.html#major-additions

Unfortunately I think we're stuck with pybind11 whilst we still need multiple inheritance: https://nanobind.readthedocs.io/en/latest/porting.html#removed-features

mfisher31, 2025-02-09:
Slightly unrelated, but is it possible to utilize robotpy from a C++ project?  e.g. from a c++ project provide pre-built python modules available, and from our code use pybind11 directly to manage the context & create new bindings where needed.  This one option 9431 is considering for a hybrid c++/python firmware this year.

Anyhow, this type of usage in theory would have HAL loading exactly as fast as a c++ native firmware.

virtuald, 2025-02-09:
Feel free to create a separate discussion about that topic, but yes in theory it's possible.