robotpy/mostrobotpy #38: Adding Auto-Wi-fi Switching When Deploying Robot Code

PhantomLel, 2023-11-16:
Hi guys, I realize that this is not at all a bug or an issue with RobotPy, but would it be possible to automatically switch to the robot network when running the deploy command? I found it to be a mild annoyance to have to switch all the time and I think it would be cool to have the program automatically take care of that for you. I would love to add this feature myself, could I get some guidance on where to start?

virtuald, 2023-11-17:
I suspect this might be challenging to implement correctly for all platforms that WPILib supports (Windows, Linux, OSX), and it would be even more challenging to implement solely in python -- I suspect there's a lot of platform-specific code that would need to run, and might change the deploy to require administrative privileges depending on the platform. Additionally, it would probably be better to do this for all FRC languages, not just for RobotPy? 

Have you looked to see what is required to programmatically change the wireless settings on Windows/Linux/OSX?

virtuald, 2023-11-28:
I'm not going to personally work on this, and I think this would be more appropriate as a general GradleRIO/wpilib feature and should be developed there. Happy to discuss more if you have other ideas.

closed 2023-11-28 by virtuald