robotpy/mostrobotpy #45: ntcore sometimes returns list[int] instead of bytes?

auscompgeek, 2023-12-17:
Mentioned in passing by Photonvision folks, but I don't know any of the details.

> Inherited from how NT4 returns the raw bytes topic value

_Originally posted by @gerth2 in https://github.com/PhotonVision/photonvision/pull/1040#discussion_r1423366787_

virtuald, 2023-12-17:
I just tried it by modifying the simple_*.py examples to getRaw/setRaw and this isn't the case. Closing unless there's a reproducer.

gerth2, 2023-12-17:
[image]
 ntcore 2024.0.0b3.post1 - `TimestampedRaw` is typehinted to lists of ints, not bytes

gerth2, 2023-12-17:
[image]
At runtime on a roboRIO - looks like the type does indeed come back as int - 
[image]

commit 42328234 (2023-12-17): Add type_caster for ntcore raw values

- std::vector<uint8_t> is only used in raw contexts, so the type caster
  intercepts and converts the value automatically to/from bytes
- Fixes #45
- There may be other places we could use this technique?
commit 53490988 (2023-12-18): Add type_caster for ntcore raw values

- std::vector<uint8_t> is only used in raw contexts, so the type caster
  intercepts and converts the value automatically to/from bytes
- Fixes #45
- There may be other places we could use this technique?
commit 80147082 (2023-12-18): Add type_caster for ntcore raw values

- std::vector<uint8_t> is only used in raw contexts, so the type caster
  intercepts and converts the value automatically to/from bytes
- Fixes #45
- There may be other places we could use this technique?