robotpy/mostrobotpy #47: Add protobuf serialization support for ntcore

virtuald, 2023-12-21:
Not really sure if anyone is going to care about this? Seems likely a bit complicated, and we'd need to build the protobuf wheel in roborio-wheels.

closed 2026-01-02 by virtuald

zachwaffle4, 2026-10-05:
given that it's been almost 3 years since this issue, it might be worth revisiting because the way wpilib c++ does protobufs is different now, and some major things like commands v3 use protobufs for telemetry

virtuald, 2026-10-06:
Well, nobody has implemented commands v3 for Python so that's one less thing to worry about ;)

I think the complicated parts would be related to the C++/protobuf/Python interop layer that would be required, and I don't think that complexity has gone away. I don't see much upside to implementing protobuf support. Feel free to produce a demo that proves me wrong.

zachwaffle4, 2026-10-06:
> Well, nobody has implemented commands v3 for Python so that's one less thing to worry about ;)

introducing https://github.com/zachwaffle4/commandsv3-python

virtuald, 2026-10-06:
> introducing https://github.com/zachwaffle4/commandsv3-python

Neat.

auscompgeek, 2026-11-12:
If WPILib C++ used the standard C++ protobuf libraries we could just use pybind11_protobuf and happy days. We're unfortunately not in that world, but there's no reason now that we couldn't use the Python protobuf library.