robotpy/mostrobotpy #56: [BUG]: getDeployDirectory incompatible with simulation

LandonBayer, 2024-01-07:
### Problem description

Hello,

When testing simulation on our base code, a teammate and I had an issue in which pathplanner went to the wrong directory for paths due to the way it is traced (sorry I haven't written these before). This issue is pretty similar to https://github.com/robotpy/mostrobotpy/issues/31 , and at the end, y'all responded saying you're open to suggestions. My humble idea is to instead use the getOperatingDirectory function (https://robotpy.readthedocs.io/projects/robotpy/en/stable/wpilib/functions.html#wpilib.getOperatingDirectory), which returns a usable output on both a regular computer and the roboRIO as depicted in the docs. Unless I'm missing something, this should bridge the gap between the simulator and real robot so that there's no confusion.

Our temporary solution was to move the auto paths into .venv/bin so that the path is found, but obviously that isn't very good.

Link to our code: https://github.com/1757WestwoodRobotics/RobotBase 

The specific section (robotcontainer line ~75):
```bash
pathsPath = os.path.join(wpilib.getDeployDirectory(), "pathplanner", "autos") #as of right now, it is necessary to move the deploy folder into .venv/scripts or .venv/bin
        for file in os.listdir(pathsPath):
            relevantName = file.split(".")[0]
            auton = PathPlannerAuto(relevantName)
            wpilib.SmartDashboard.putData(f"autos/{relevantName}", auton)
            self.chooser.addOption(relevantName, auton)
```
Error message:
```bash
FileNotFoundError: [Errno 2] No such file or directory: '/home/landon/code/frc/robotcode/RobotBase/.venv/bin/deploy/pathplanner/autos/New Auto.auto'
#real path should be ..../RobotBase/deploy/pathplanner/autos/New Auto.auto
```
According to the getDeployDirectory docs, when simulating it will instead run dirname(robot.py)/deplay, however according to this behavior it either isn't doing that or isn't detecting simulation in some way (maybe due to the new way it runs with python -m robotpy sim instead of python robot.py sim?)

Thanks for your help, and feel free to ask for any questions or greater clarification

### Operating System

Linux

### Installed Python Packages

```text
astroid                  3.0.2
bcrypt                   4.1.2
black                    23.12.1
cffi                     1.16.0
click                    8.1.7
cryptography             41.0.7
dill                     0.3.7
exceptiongroup           1.2.0
iniconfig                2.0.0
isort                    5.13.2
mccabe                   0.7.0
mypy-extensions          1.0.0
packaging                23.2
paramiko                 3.4.0
pathspec                 0.12.1
phoenix6                 24.0.0b5
photonlibpy              2024.1.1b42
Pint                     0.23
pip                      22.0.2
platformdirs             4.1.0
pluggy                   1.3.0
pycparser                2.21
pyfrc                    2024.0.0b3
pylint                   3.0.3
PyNaCl                   1.5.0
pynetconsole             2.0.4
pyntcore                 2024.0.0b4
pytest                   7.4.4
pytest-reraise           2.1.2
robotpy                  2024.0.0b4
robotpy-apriltag         2024.0.0b4
robotpy-cli              2024.0.0
robotpy-commands-v2      2024.0.0b3
robotpy-cscore           2024.0.0b4
robotpy-ctre             2024.0.0b4
robotpy-hal              2024.0.0b4
robotpy-halsim-ds-socket 2024.0.0b4
robotpy-halsim-gui       2024.0.0b4
robotpy-halsim-ws        2024.0.0b4
robotpy-installer        2024.0.2
robotpy-navx             2024.0.0b1
robotpy-pathplannerlib   2024.0.0b6.post3
robotpy-rev              2024.0.0b1.post1
robotpy-wpilib-utilities 2024.0.0b1
robotpy-wpimath          2024.0.0b4
robotpy-wpinet           2024.0.0b4
robotpy-wpiutil          2024.0.0b4
setuptools               69.0.3
tomli                    2.0.1
tomlkit                  0.12.3
typing_extensions        4.9.0
wpilib                   2024.0.0b4
```

### Reproducible example code

```text
Sorry it's not minimum but using https://github.com/1757WestwoodRobotics/RobotBase will work
```

virtuald, 2024-01-07:
Please upgrade to the kickoff release, I believe the issue should be addressed there.

closed 2024-01-07 by virtuald

virtuald, 2024-01-07:
Also, I note that you're using commands2. If you install it separately, the beta release will continue to work with the kickoff release by adding `robotpy-commands-v2==2024.0.0b4` to `tool.robotpy.requires` in `pyproject.toml` (new for 2024 kickoff release) -- I'm just not comfortable declaring a kickoff release for it yet.