robotpy/mostrobotpy #60: [BUG]: NT struct array topic doesn't publish schema

truher, 2024-01-09:
### Problem description

```
x = ntcore.NetworkTableInstance.getDefault().getStructArrayTopic("foo", Translation2d).publish()
x.set([Translation2d(0,0)])
```
is rendered as  opaque in glass, and there is only one publisher, "foo."

adding 
```
y = ntcore.NetworkTableInstance.getDefault().getStructTopic("foo", Translation2d).publish()
y.set(Translation2d(0,0)
```
adds the "/.schema" publisher and fixes the array rendering.

### Operating System

Linux

virtuald, 2024-01-24:
WPILib fix in https://github.com/wpilibsuite/allwpilib/pull/6303

auscompgeek, 2024-04-07:
Fixed in v2024.3.1

closed 2024-04-07 by auscompgeek