robotpy/mostrobotpy #65: [BUG]: ADIS16470_IMU crashes in test

etiennebeaulac, 2024-01-20:
### Problem description

When we run `python -m robotpy test` with an `ADIS16470_IMU`, it crashes.

```
C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\python.exe -m robotpy test 
11:09:19:364 WARNING : pyfrc.physics       : Cannot enable physics support, C:\Users\etienne.beaulac\Desktop\test\physics.py not found
============================= test session starts =============================
platform win32 -- Python 3.12.1, pytest-7.4.4, pluggy-1.3.0
rootdir: C:\Users\etienne.beaulac\Desktop\test\tests
plugins: reraise-2.1.2
collected 4 items

pyfrc_test.py .Windows fatal exception: access violation

Thread 0x00003ca8 (most recent call first):
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\ntcore\_logutil.py", line 94 in _logging_thread
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\threading.py", line 1010 in run
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\threading.py", line 1073 in _bootstrap_inner
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\threading.py", line 1030 in _bootstrap

Current thread 0x00001a38 (most recent call first):
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\pyfrc\test_support\pytest_plugin.py", line 114 in robot
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\fixtures.py", line 911 in _teardown_yield_fixture
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\fixtures.py", line 1024 in finish
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\fixtures.py", line 701 in <lambda>
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\runner.py", line 526 in teardown_exact
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\runner.py", line 182 in pytest_runtest_teardown
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\pluggy\_callers.py", line 77 in _multicall
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\pluggy\_manager.py", line 115 in _hookexec
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\pluggy\_hooks.py", line 493 in __call__
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\runner.py", line 262 in <lambda>
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\runner.py", line 341 in from_call
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\runner.py", line 261 in call_runtest_hook
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\runner.py", line 222 in call_and_report
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\runner.py", line 134 in runtestprotocol
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\runner.py", line 114 in pytest_runtest_protocol
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\pluggy\_callers.py", line 77 in _multicall
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\pluggy\_manager.py", line 115 in _hookexec
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\pluggy\_hooks.py", line 493 in __call__
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\main.py", line 350 in pytest_runtestloop
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\pluggy\_callers.py", line 77 in _multicall
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\pluggy\_manager.py", line 115 in _hookexec
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\pluggy\_hooks.py", line 493 in __call__
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\main.py", line 325 in _main
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\main.py", line 271 in wrap_session
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\main.py", line 318 in pytest_cmdline_main
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\pluggy\_callers.py", line 77 in _multicall
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\pluggy\_manager.py", line 115 in _hookexec
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\pluggy\_hooks.py", line 493 in __call__
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\_pytest\config\__init__.py", line 169 in main
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\pyfrc\mains\cli_test.py", line 115 in _run_test
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\pyfrc\mains\cli_test.py", line 62 in run
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\robotpy\main.py", line 314 in main
  File "C:\Users\etienne.beaulac\AppData\Local\miniconda3\envs\frc2024\Lib\site-packages\robotpy\__main__.py", line 4 in <module>
  File "<frozen runpy>", line 88 in _run_code
  File "<frozen runpy>", line 198 in _run_module_as_main

Process finished with exit code -1073741819 (0xC0000005)
```

### Operating System

Windows

### Installed Python Packages

```text
bcrypt                   4.1.2     
cffi                     1.16.0    
colorama                 0.4.6     
cryptography             41.0.7    
iniconfig                2.0.0     
numpy                    1.26.3    
packaging                23.2      
paramiko                 3.4.0     
Pint                     0.23      
pip                      23.3.2
pluggy                   1.3.0
pycparser                2.21
pyfrc                    2024.0.1
PyNaCl                   1.5.0
pynetconsole             2.0.4
pyntcore                 2024.1.1.1
pytest                   7.4.4
pytest-reraise           2.1.2
robotpy                  2024.1.1.3
robotpy-apriltag         2024.1.1.1
robotpy-cli              2024.0.0
robotpy-commands-v2      2024.0.0b4
robotpy-cscore           2024.1.1.1
robotpy-hal              2024.1.1.1
robotpy-halsim-ds-socket 2024.1.1.1
robotpy-halsim-gui       2024.1.1.1
robotpy-halsim-ws        2024.1.1.1
robotpy-installer        2024.1.1
robotpy-rev              2024.2.0
robotpy-wpilib-utilities 2024.0.0
robotpy-wpimath          2024.1.1.1
robotpy-wpinet           2024.1.1.1
robotpy-wpiutil          2024.1.1.1
setuptools               69.0.3
tomli                    2.0.1
typing_extensions        4.9.0
wheel                    0.42.0
wpilib                   2024.1.1.1
```

### Reproducible example code

```text
import wpilib

class Robot(wpilib.TimedRobot):
    def robotInit(self):
        self.gyro = wpilib.ADIS16470_IMU()
```

virtuald, 2024-01-20:
Partial stack trace on Linux shows it dying in the destructor:

```
(gdb) bt
#0  frc::ADIS16470_IMU::Close (this=this@entry=0x7fffd4001990) at /work/wpilibc/src/main/native/cpp/ADIS16470_IMU.cpp:458
#1  0x00007fffe733b6bb in frc::ADIS16470_IMU::~ADIS16470_IMU (this=0x7fffd4001990, __in_chrg=<optimized out>) at /work/wpilibc/src/main/native/cpp/ADIS16470_IMU.cpp:486
#2  0x00007fffe733b6fd in frc::ADIS16470_IMU::~ADIS16470_IMU (this=0x7fffd4001990, __in_chrg=<optimized out>) at /work/wpilibc/src/main/native/cpp/ADIS16470_IMU.cpp:487
```

A new WPILib release is rolling out soon, and looks like there are changes to the 16470 so I'll take a look then.

virtuald, 2024-01-21:
This was not fixed in the 2024.2 WPILib release.

commit a07b9d97 (2024-01-21): Add test to check if ADIS16470 crashes

- Fixes #65
closed 2024-01-21 by virtuald