robotpy/mostrobotpy #81: Add pyi to linuxarm64 wheels

TheTripleV, 2024-02-18:
Unlike the rio wheels, editing code directly on a Raspberry Pi with intellisense is possible (Vscode Remote).

It would be good to run stubgen on wheels targeting the Pi.

virtuald, 2024-11-03:
To do this, someone would need to add qemu-arm-static to the various docker images and check that it can run the python interpreter that we cross-compile. Should be straightforward?

auscompgeek, 2025-01-26:
GitHub Actions provide native Linux ARM runners now, which hopefully makes this even more straightforward. Gotta get the build to work in a Debian Bookworm (at the time of writing), which [I haven't managed yet](https://github.com/auscompgeek/mostrobotpy/actions/workflows/dist.yml?query=branch%3Aci-linux-aarch64).

PR #153 (2025-02-08): Build Linux AArch64 wheels natively
This adds jobs to build wheels on Debian Bookworm native ARM64, with the intent of publishing these wheels to PyPI.

Closes #81
closed 2025-02-08 by virtuald