robotpy/mostrobotpy #90: Missing hal.simulation functions for registering sim callbacks

bhall-ctre, 2024-04-30:
### Problem description

Some of the `hal.simulation` functions for registering callbacks are missing. For example, there is no `registerSimValueChangedCallback`, even though there is a `cancelSimValueChangedCallback`. These callbacks are necessary for vendors to use WPILib sim GUI inputs.

I believe all the relevant callback APIs are in SimDeviceData.h: https://github.com/wpilibsuite/allwpilib/blob/main/hal/src/main/native/include/hal/simulation/SimDeviceData.h

### Operating System

Windows

### Installed Python Packages

```text
pyfrc                          2024.0.1
pyntcore                       2024.3.2.0
robotpy                        2024.3.2.1
robotpy-cli                    2024.0.0
robotpy-commands-v2            2024.3.1
robotpy-hal                    2024.3.2.0
robotpy-halsim-gui             2024.3.2.0
robotpy-installer              2024.2.2
robotpy-wpilib-utilities       2024.0.0
robotpy-wpimath                2024.3.2.0
robotpy-wpinet                 2024.3.2.0
robotpy-wpiutil                2024.3.2.0
wpilib                         2024.3.2.0
```

TheTripleV, 2024-05-01:
https://github.com/robotpy/mostrobotpy/blob/main/subprojects/robotpy-hal/gen/simulation/SimDeviceData.yml

They are all manually ignored right now for some reason.

virtuald, 2024-05-01:
Nobody ever needed them so I never bothered. It shouldn't be that bad. Which specific callbacks do you need?

commit e1f5820a (2024-05-03): Add registerSimValue callbacks

- Fixes #90
PR #91 (2024-05-04): Add registerSimValue callbacks
- Fixes #90

@bhall-ctre please try this out and let me know if this works for you -- it should, I even added tests! If you don't want to build it yourself, you should be able to download the built wheels from the github actions job associated with this PR.
closed 2024-05-04 by virtuald