robotpy/mostrobotpy #103: mypy strict doesn't like make_wpistruct

truher, 2024-10-10:
I think the default None for the cls argument is the issue.  What does it mean for cls to be None?

https://github.com/robotpy/mostrobotpy/blob/a614f97d5eb790ddab2a9e7815dfe2179ac8b1ee/subprojects/robotpy-wpiutil/wpiutil/wpistruct/dataclass.py#L42

```
Type of "make_wpistruct" is partially unknown
  Type of "make_wpistruct" is "(cls: Unknown | None = None, /, *, name: str | None = None) -> (((cls: Unknown) -> Unknown) | Unknown)"Pylance[reportUnknownMemberType](https://github.com/microsoft/pyright/blob/main/docs/configuration.md#reportUnknownMemberType)
Untyped class decorator obscures type of class; ignoring decoratorPylance[reportUntypedClassDecorator](https://github.com/microsoft/pyright/blob/main/docs/configuration.md#reportUntypedClassDecorator)
```

virtuald, 2024-10-10:
It's a class decorator, and I think that's the standard way to write a decorator? There's probably an example somewhere how one should properly annotate class decorators.

truher, 2024-10-10:
i see.  ok, thanks.