robotpy/mostrobotpy #104: [BUG]: Missing functions with DCMotorSim

daltzctr, 2024-10-11:
### Problem description

WPILib has various getPosition and getVelocity in specified units: https://github.wpilib.org/allwpilib/docs/development/java/edu/wpi/first/wpilibj/simulation/DCMotorSim.html#getAngularVelocity()

but RobotPy is missing `getAngularPositionRotations`, `getAngularVelocityRotations`, and maybe other functions: https://robotpy.readthedocs.io/projects/robotpy/en/stable/wpilib.simulation/DCMotorSim.html#wpilib.simulation.DCMotorSim.getAngularVelocity

### Operating System

Windows, MacOS, Linux, RoboRIO, Raspbian

virtuald, 2024-10-11:
Those functions do not exist in C++: https://github.com/wpilibsuite/allwpilib/blob/28cb7cf757e8dce89aab6d8181203b665afca62a/wpilibc/src/main/native/include/frc/simulation/DCMotorSim.h#L60

My guess is it's just a unit conversion, so C++ doesn't need them? If you can provide which units you need we can add the corresponding functions.

virtuald, 2024-12-31:
There's missing setter functions, fixed in https://github.com/wpilibsuite/allwpilib/pull/7613

commit 06e2b0f8 (2025-01-06): Add compatibility methods to DCMotorSim

- Fixes #104
PR #128 (2025-01-07): Add compatibility methods to DCMotorSim
- Fixes #104
closed 2025-01-07 by virtuald