robotpy/mostrobotpy #117: [BUG]: DataLogManager type hints do not indicate it can't be constructed

TheTripleV, 2024-12-19:
### Problem description

```py
import wpilib
class Robot(wpilib.TimedRobot):
    def __init__(self, period = 0.02):
        super().__init__()
        wpilib.DataLogManager()
```
`wpilib.DataLogManager()` can't be constructed but there's no type error.

The pyi for DataLogManager needs a `__init__: None` in it

### Operating System

Windows

virtuald, 2024-12-31:
This feels like a stubgen bug?

auscompgeek, 2025-01-01:
Is it even possible to introspect the class and see there's no constructor? AFAICT from Python it _looks_ like `__init__` is defined. It even exists in `wpilib.DataLogManager.__dict__`.