robotpy/mostrobotpy #118: [BUG]: 2025.0.0b3 Deploy

halfaleague, 2024-12-23:
### Problem description

I get this error when trying to run `robotpy deploy`
(I ran sync -- see contents of pip cache below)
```
Collecting robotpy==2025.0.0b3
  Downloading http://localhost:45953/pip_cache/robotpy-2025.0.0b3-py3-none-any.whl (2.1 kB)
ERROR: Could not find a version that satisfies the requirement wpilib (from versions: none)
ERROR: No matching distribution found for wpilib
ERROR: installing packages: Command '/home/admin/rpip --no-cache-dir --disable-pip-version-check install --no-index --root-user-action=ignore --find-links http://localhost:45953/pip_cache/ --upgrade --upgrade-strategy=eager robotpy==2025.0.0b3 wpilib' returned non-zero error status 1

If 'no matching distribution found', run 'python -m robotpy sync' to download your
project requirements from the internet (or --no-install to ignore).
```

Here is the contents of ~/wpilib/2025/robotpy/pip_cache
Notice the difference between symlinks and the source files -- the wheel files install if the symlinks are used.
I added the symlinks to see if the version is preventing it from installing (made the links like robotpy).
```
$ ll
total 18M
-rw-rw-r-- 1 luke luke 1.8M Dec 23 15:18 pyntcore-2025.0.0b3-cp313-cp313-linux_roborio.whl
lrwxrwxrwx 1 luke luke   49 Dec 23 15:20 pyntcore-2025.0.0b3-py3-none-any.whl -> pyntcore-2025.0.0b3-cp313-cp313-linux_roborio.whl
-rw-rw-r-- 1 luke luke 2.1K Dec 23 15:18 robotpy-2025.0.0b3-py3-none-any.whl
-rw-rw-r-- 1 luke luke 8.7K Dec 23 15:18 robotpy_cli-2024.0.0-py3-none-any.whl
-rw-rw-r-- 1 luke luke 1.2M Dec 23 15:18 robotpy_hal-2025.0.0b3-cp313-cp313-linux_roborio.whl
lrwxrwxrwx 1 luke luke   52 Dec 23 15:20 robotpy_hal-2025.0.0b3-py3-none-any.whl -> robotpy_hal-2025.0.0b3-cp313-cp313-linux_roborio.whl
-rw-rw-r-- 1 luke luke  48K Dec 23 15:18 robotpy_wpilib_utilities-2025.0.0b2-py3-none-any.whl
-rw-rw-r-- 1 luke luke 6.0M Dec 23 15:18 robotpy_wpimath-2025.0.0b3-cp313-cp313-linux_roborio.whl
lrwxrwxrwx 1 luke luke   56 Dec 23 15:20 robotpy_wpimath-2025.0.0b3-py3-none-any.whl -> robotpy_wpimath-2025.0.0b3-cp313-cp313-linux_roborio.whl
-rw-rw-r-- 1 luke luke 692K Dec 23 15:18 robotpy_wpinet-2025.0.0b3-cp313-cp313-linux_roborio.whl
lrwxrwxrwx 1 luke luke   55 Dec 23 15:20 robotpy_wpinet-2025.0.0b3-py3-none-any.whl -> robotpy_wpinet-2025.0.0b3-cp313-cp313-linux_roborio.whl
-rw-rw-r-- 1 luke luke 3.1M Dec 23 15:18 robotpy_wpiutil-2025.0.0b3-cp313-cp313-linux_roborio.whl
lrwxrwxrwx 1 luke luke   56 Dec 23 15:20 robotpy_wpiutil-2025.0.0b3-py3-none-any.whl -> robotpy_wpiutil-2025.0.0b3-cp313-cp313-linux_roborio.whl
-rw-rw-r-- 1 luke luke 4.6M Dec 23 15:18 wpilib-2025.0.0b3-cp313-cp313-linux_roborio.whl
lrwxrwxrwx 1 luke luke   47 Dec 23 15:20 wpilib-2025.0.0b3-py3-none-any.whl -> wpilib-2025.0.0b3-cp313-cp313-linux_roborio.whl
```

But once the symlinks is used to hackily install them -- the symbol `PyThreadState_GetUnchecked` seems new to Python 3.13: 
https://docs.python.org/3/c-api/init.html#c.PyThreadState_GetUnchecked
(the roborio is running python 3.12)

```
admin@roboRIO-7660-FRC:/usr/local/lib/python3.12/site-packages# python3
Python 3.12.1 (main, Dec 21 2023, 22:22:43) [GCC 12.1.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import wpiutil
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.12/site-packages/wpiutil/__init__.py", line 4, in <module>
    from ._wpiutil import (
ImportError: /usr/local/lib/python3.12/site-packages/wpiutil/_wpiutil.so: undefined symbol: PyThreadState_GetUnchecked
```

### Operating System

Linux

### Installed Python Packages

```text
$ robotpy installer list  
15:25:49:001 INFO    : robotpy.installer   : RobotPy Installer 2025.0.0b3
15:25:49:001 INFO    : robotpy.installer   : -> caching files at /home/luke/wpilib/2025/robotpy
15:25:49:001 INFO    : robotpy.installer   : Settings loaded from /home/luke/code/basic-robotpy/.wpilib/wpilib_preferences.json
15:25:49:001 INFO    : robotpy.installer   : Finding robot for team 7660
15:25:49:006 INFO    : robotpy.installer   : -> Robot is at 10.76.60.2
15:25:49:006 INFO    : robotpy.installer   : Connecting to robot via SSH at 10.76.60.2
15:25:49:168 INFO    : paramiko.transport  : Connected (version 2.0, client OpenSSH_8.3)
15:25:49:303 INFO    : paramiko.transport  : Auth banner: b'NI Linux Real-Time (run mode)\n\nLog in with your NI-Auth credentials.\n\n'
15:25:49:304 INFO    : paramiko.transport  : Authentication (password) successful!
15:25:50:364 INFO    : robotpy.installer   : -> RoboRIO image version: 2025_v1.1
15:25:53:514 INFO    : paramiko.transport.sftp: [chan 2] Opened sftp connection (server version 3)
15:25:53:528 INFO    : paramiko.transport.sftp: [chan 2] sftp session closed.
Package                  Version
------------------------ ----------
pip                      23.2.1
pyntcore                 2025.0.0b3
robotpy                  2025.0.0b3
robotpy-cli              2024.0.0
robotpy-hal              2025.0.0b3
robotpy-wpilib-utilities 2025.0.0b2
robotpy-wpimath          2025.0.0b3
robotpy-wpinet           2025.0.0b3
robotpy-wpiutil          2025.0.0b3
wpilib                   2025.0.0b3
```

### Reproducible example code

```text
Any basic example python code that has an empty robot class...
```

auscompgeek, 2024-12-23:
How did you manage to install Python 3.12 on the RIO 2025 beta image? We've only built Python 3.13 for 2025.

virtuald, 2024-12-23:
You can technically install the 2024 python on the new image if you try hard enough, but in theory robotpy-installer shouldn't let you do it... oh, maybe if you do --ignore-image-version?

In any case, you should try to do `robotpy installer uninstall-robotpy` and then try again.

virtuald, 2024-12-23:
Oh, and do `rm -rf /home/luke/wpilib/2025/robotpy`, those symlinks are going to cause problems.

halfaleague, 2024-12-23:
The roborio hasn't been modified other than official tooling. (deploy)
Java 2025 beta examples run fine.

Where is the official 2025 beta image from?
I imaged using this download -- linked from some Chief Delphi forum:
https://github.com/wpilibsuite/2025Beta/releases/tag/NI_GAME_TOOLS_BETA_2

[image]

And the `robotpy installer uninstall-robotpy` didn't work -- Python 3.12 is still there -- I think it comes with the image, yes?

Thanks!

virtuald, 2024-12-24:
Python does not come with the image, the installer automatically grabs it as part of the sync process and installs it.

If you SSH into the robot, and run `opkg remove python312` that will remove the old version. Or you can reimage.

halfaleague, 2024-12-26:
Ran the following and re-ran / sync / deploy -- everything seems to work now, thanks!

```
NI Linux Real-Time (run mode)

Log in with your NI-Auth credentials.

admin@roboRIO-7660-FRC:~# opkg remove python312
Removing python312 (3.12.1) from root...
```

Just for posterity -- I had assumed / read somewhere that 3.12 was the version so I used that in my local environment to run `sync/deploy`.
The deploy command must use your local version to decide what to install remotely **but** that may not match up with the version the packages are built with.

closed 2024-12-26 by halfaleague

virtuald, 2024-12-26:
robotpy-installer only supports installing a single version of python, so my theory is that you used the 2024 installer to install to the 2025 image. If we implement https://github.com/robotpy/robotpy-installer/issues/125 and https://github.com/robotpy/robotpy-installer/issues/126 then hopefully this won't occur in the future.

Glad it's working for you!