robotpy/mostrobotpy #140: [BUG]: wpilib.Alert does not show in SmartDashboard

etiennebeaulac, 2025-01-18:
### Problem description

The new Persistent Alerts of WPILIb should display alerts in Network Tables (table SmartDashboard), but it does not seem to work.

### Operating System

Windows

### Installed Python Packages

```text
ast-selector             0.2.0
asttokens                2.4.1
bcrypt                   4.2.1
black                    24.1.1
cffi                     1.17.1
click                    8.1.8
colorama                 0.4.6
coverage                 7.4.1
cryptography             44.0.0
flexcache                0.3
flexparser               0.4
iniconfig                2.0.0
msvc_runtime             14.42.34433
mypy-extensions          1.0.0
numpy                    2.2.1
opencv-python            4.10.0.84
packaging                23.2
paramiko                 3.5.0
pathspec                 0.12.1
photonlibpy              2025.0.0b8
Pint                     0.24.4
pip                      24.2
platformdirs             4.3.6
pluggy                   1.5.0
pycparser                2.22
pyfrc                    2025.0.0
PyNaCl                   1.5.0
pynetconsole             2.0.4
pyntcore                 2025.2.1
pytest                   8.3.4
pytest-reraise           2.1.2
robotpy                  2025.2.1.0
robotpy-apriltag         2025.2.1
robotpy-cli              2024.0.0
robotpy-commands-v2      2025.1.1
robotpy-cscore           2025.2.1
robotpy-hal              2025.2.1
robotpy-halsim-ds-socket 2025.2.1
robotpy-halsim-gui       2025.2.1
robotpy-halsim-ws        2025.2.1
robotpy-installer        2025.0.0
robotpy-rev              2025.0.1
robotpy-wpilib-utilities 2025.0.0
robotpy-wpimath          2025.2.1
robotpy-wpinet           2025.2.1
robotpy-wpiutil          2025.2.1
setuptools               75.1.0
six                      1.17.0
tomli                    2.2.1
tomlkit                  0.13.2
typing_extensions        4.12.2
wheel                    0.44.0
wpilib                   2025.2.1
```

### Reproducible example code

```text
import wpilib
from wpilib import TimedRobot

class Robot(TimedRobot):
    def robotInit(self):
        self.alert = wpilib.Alert("Test alert", wpilib.Alert.AlertType.kError)
        self.alert.set(True)
```

virtuald, 2025-01-18:
I agree that it doesn't seem to work.

virtuald, 2025-01-18:
Well. They sorta work, but not in a robot context for some reason. I'll have to dig into it later.

virtuald, 2025-01-19:
It doesn't work in a C++ robot project either, there must be some initialization or something that is missing.

virtuald, 2025-01-19:
This is a bug in the C++ library, fix should make its way into the next release.

virtuald, 2025-01-19:
As a workaround, if you create multiple alerts then it will work.

commit fbb5b110 (2025-02-16): wpilib: Add context support and tests to Alert

- Fixes #140
commit e52deb09 (2025-02-16): wpilib: Add context support and tests to Alert

- Fixes #140
PR #157 (2025-02-16): wpilib: Add context support and tests to Alert
- Fixes #140
closed 2025-02-16 by virtuald