robotpy/mostrobotpy #149: Memory issues

virtuald, 2025-02-04:
RoboRIO 1 with current config. Here's python via pmap:

```
396:   python3
00010000      4K r-x-- python3.13
00020000      4K r---- python3.13
00021000      4K rw--- python3.13
0111d000   1500K rw---   [ anon ]
b5f69000    108K r-x-- libncurses.so.5.9
b5f84000     64K ----- libncurses.so.5.9
b5f94000      4K r---- libncurses.so.5.9
b5f95000      4K rw--- libncurses.so.5.9
b5f96000     16K r-x-- _random.cpython-313-arm-linux-gnueabi.so
b5f9a000     60K ----- _random.cpython-313-arm-linux-gnueabi.so
b5fa9000      4K r---- _random.cpython-313-arm-linux-gnueabi.so
b5faa000      4K rw--- _random.cpython-313-arm-linux-gnueabi.so
b5fab000     16K r-x-- _bisect.cpython-313-arm-linux-gnueabi.so
b5faf000     60K ----- _bisect.cpython-313-arm-linux-gnueabi.so
b5fbe000      4K r---- _bisect.cpython-313-arm-linux-gnueabi.so
b5fbf000      4K rw--- _bisect.cpython-313-arm-linux-gnueabi.so
b5fc0000    492K rw---   [ anon ]
b603b000     80K r-x-- math.cpython-313-arm-linux-gnueabi.so
b604f000     60K ----- math.cpython-313-arm-linux-gnueabi.so
b605e000      4K r---- math.cpython-313-arm-linux-gnueabi.so
b605f000      4K rw--- math.cpython-313-arm-linux-gnueabi.so
b6060000     16K r-x-- _posixsubprocess.cpython-313-arm-linux-gnueabi.so
b6064000     60K ----- _posixsubprocess.cpython-313-arm-linux-gnueabi.so
b6073000      4K r---- _posixsubprocess.cpython-313-arm-linux-gnueabi.so
b6074000      4K rw--- _posixsubprocess.cpython-313-arm-linux-gnueabi.so
b6075000    256K rw---   [ anon ]
b60b5000    124K r-x-- liblzma.so.5.2.3
b60d4000     60K ----- liblzma.so.5.2.3
b60e3000      4K r---- liblzma.so.5.2.3
b60e4000      4K rw--- liblzma.so.5.2.3
b60e5000     28K r-x-- _lzma.cpython-313-arm-linux-gnueabi.so
b60ec000     60K ----- _lzma.cpython-313-arm-linux-gnueabi.so
b60fb000      4K r---- _lzma.cpython-313-arm-linux-gnueabi.so
b60fc000      4K rw--- _lzma.cpython-313-arm-linux-gnueabi.so
b60fd000     56K r-x-- libbz2.so.1.0.6
b610b000     60K ----- libbz2.so.1.0.6
b611a000      4K r---- libbz2.so.1.0.6
b611b000      4K rw--- libbz2.so.1.0.6
b611c000     16K r-x-- _bz2.cpython-313-arm-linux-gnueabi.so
b6120000     60K ----- _bz2.cpython-313-arm-linux-gnueabi.so
b612f000      4K r---- _bz2.cpython-313-arm-linux-gnueabi.so
b6130000      4K rw--- _bz2.cpython-313-arm-linux-gnueabi.so
b6131000     76K r-x-- libz.so.1.2.11
b6144000     60K ----- libz.so.1.2.11
b6153000      4K r---- libz.so.1.2.11
b6154000      4K rw--- libz.so.1.2.11
b6155000     32K r-x-- zlib.cpython-313-arm-linux-gnueabi.so
b615d000     60K ----- zlib.cpython-313-arm-linux-gnueabi.so
b616c000      4K r---- zlib.cpython-313-arm-linux-gnueabi.so
b616d000      4K rw--- zlib.cpython-313-arm-linux-gnueabi.so
b616e000    108K r-x-- libgcc_s.so.1
b6189000     60K ----- libgcc_s.so.1
b6198000      4K r---- libgcc_s.so.1
b6199000      4K rw--- libgcc_s.so.1
b619a000     24K r-x-- libffi.so.6.0.4
b61a0000     60K ----- libffi.so.6.0.4
b61af000      4K r---- libffi.so.6.0.4
b61b0000      4K rw--- libffi.so.6.0.4
b61b1000    104K r-x-- _ctypes.cpython-313-arm-linux-gnueabi.so
b61cb000     60K ----- _ctypes.cpython-313-arm-linux-gnueabi.so
b61da000      4K r---- _ctypes.cpython-313-arm-linux-gnueabi.so
b61db000      8K rw--- _ctypes.cpython-313-arm-linux-gnueabi.so
b61dd000    256K rw---   [ anon ]
b621d000     16K r-x-- fcntl.cpython-313-arm-linux-gnueabi.so
b6221000     60K ----- fcntl.cpython-313-arm-linux-gnueabi.so
b6230000      4K r---- fcntl.cpython-313-arm-linux-gnueabi.so
[... 186 lines cut ...]
+b371c000   1724K r-x-- libstdc++.so.6.0.30
+b38cb000     60K ----- libstdc++.so.6.0.30
+b38da000     20K r---- libstdc++.so.6.0.30
+b38df000      8K rw--- libstdc++.so.6.0.30
+b38e1000      8K rw---   [ anon ]
+b38e3000     16K r-x-- libatomic.so.1.2.0
+b38e7000     60K ----- libatomic.so.1.2.0
+b38f6000      4K r---- libatomic.so.1.2.0
+b38f7000      4K rw--- libatomic.so.1.2.0
+b38f8000      4K rw---   [ anon ]
+b38f9000   3312K r-x-- libwpiutil.so
+b3c35000     60K ----- libwpiutil.so
+b3c44000     16K r---- libwpiutil.so
+b3c48000     20K rw--- libwpiutil.so
+b3c4d000    520K rw---   [ anon ]
 b3ccf000    128K r-x-- _umath_linalg.cpython-313-arm-linux-gnueabi.so
 b3cef000     64K ----- _umath_linalg.cpython-313-arm-linux-gnueabi.so
 b3cff000      4K r---- _umath_linalg.cpython-313-arm-linux-gnueabi.so
@@ -180,4 +224,4 @@
 be8f6000    132K rwx--   [ stack ]
 bec99000      4K r-x--   [ anon ]
 ffff0000      4K r-x--   [ anon ]
- total    55636K
+ total    74084K
```

virtuald, 2025-02-04:
Currently, we recommend disabling the niwebserver via `robotpy installer niweb disable`. There are other places we could conserve memory.

* The openblas buffer is set to 8MB (and there is one per core), we could reduce it further to 4MB or 2MB, I'm not sure whether it would break anything or not
* crond on my rio takes up 8MB of RAM, in theory disabling that could be a win for low effort. It seems to only run logrotate every 5 minutes -- and I think it fails at doing that because of a config issue
* lvrt takes up a whopping 40MB of RAM, and we've talked about replacing it in the past with a C executable that does the same thing. This is mildly risky, but I've heard it can work

virtuald, 2025-02-04:
I guess a different option is to build numpy without openblas support, which would save 28MB. It's unknown whether this would decrease performance. In theory, it's supposed to accelerate multiplication.

Another alternative would be to reduce the number of threads that OpenBLAS uses:

```
>>> import os
>>> os.environ["OMP_NUM_THREADS"] = "1"
>>> import numpy
```

This reduces the amount of memory used by 10MB. Unknown performance impact.

virtuald, 2025-02-04:
I built a version of numpy that doesn't use OpenBLAS. For now, you need to explicitly enable it in your pyproject.toml:

```
[tool.robotpy]

requires = [
    "numpy==2.2.1.dev0; platform_machine == 'roborio'",
]
```

With this version of numpy, it seems to save around 15MB of RAM.

roberttuttle, 2025-02-14:
Chiming in here ... 2881 has been tracking memory issues across CD and GitHub issue threads. 
- We use roboRIO 1 (team choice on avoiding v2 if at all possible because of rampant SD card issues reported across seasons)
- We use PathPlanner, PhotonVision, rev, navx, and commands-v2
- We have followed guidance to disable NI web server to gain mem space
- We have updated toml config to use requires only and load just what we need per above
- We have updated toml config to load the non-openBLAS numpy dev version per @virtuald post above
- Current code with all the things above pushed to our main is here: https://github.com/frc2881/2025-Robot

Data point: with all of config updates above ... we have been able to sync/deploy and run full match sessions to a "fresh" roboRIO (rebooted / power-cycled) with no memory allocation exceptions/crashes. So far, so good. However, when we hit about our 5th or so deploy with a code update during dev/testing, we hit the memory limit and the deploy crashes in midflight. We then have to simply reboot the roboRIO or power-cycle and deploy fresh and all is good again for the next ~5 run/deploy cycles. Not a showstopper of course as long as we can ultimately run a match on a clean boot up at our events. It seems that something is not being GC'd and/or leaking when we deploy each time and with our current footprint it takes about 5 times to hit the wall.  Just a dev/test nuisance at this point. Hope this helps.

virtuald, 2025-02-15:
That's really interesting, there shouldn't be any residual memory usage across deploys. When deploying, there are outputs like "RoboRIO disk usage 238.2M/386.3M (62% full)" and "RoboRIO memory 203.6M/250.2M (19% full)"... are those values going up or staying the samee?

If it happens reliably, running `robotpy installer sshcmd 'ps aux'` will capture some memory related information. You can pipe it to a file, and maybe we can use that data to figure out which process is eating up more memory?

I feel like that sort of thing has been reported previously.

roberttuttle, 2025-02-15:
Will try repro and grab both the exception details and the data points on disk/mem usage to share here this weekend. There will be lots of code updates and deploys for sure heading towards week 1 event.

roberttuttle, 2025-02-18:
Following up for 2881 ... after all of the memory optimization updates, the team was able to work all weekend over many (100s) code deployments and not able to repro the memory space allocation fault observed earlier. Heading towards a week 1 event and no time to go back to methodically reverse each optimization and/or RobotPy build to trace the issue, but all seems good now.

Also, no noticeable/measurable performance issues with the custom numpy version (non OpenBLAS).

closed 2026-10-19 by virtuald