robotpy/mostrobotpy #150: Incase you wanted to build on MacOS

mikeohyan, 2025-02-04:
In case you wanted MacOS build instructions:

What you need before hand:
- Install XCode from the Apple App Store (this should get `clang` setup)
- `uv` to manage python environments (https://docs.astral.sh/uv/getting-started/installation/)

To confirm you can compile some C++ code, in a terminal you should be able to run: `clang --version`

Now to compile:
- `git clone https://github.com/robotpy/mostrobotpy.git ` to a working directory
- in that directory, run `uv init` to setup python package management
- install the package requirements: `uv add -r rdev_requirements.txt`
- install numpy `uv add numpy`
- `uv add pip`

Now to run compile command, instead of the bash script, we're going to run this instead.
In the working directory, run:
`uv run python -m devtolls ci run`

This will run for... a while, if this works you'll get a bunch of wheels (.whl) in `<working dir>/dist`

Now to use these wheels, in your python project (assuming you are using `uv` to manage packages):
Run `uv pip install <whl path>` for whatever you need.
There are packages that will have dependencies, if asked, install what's missing first.
Some packages are ok with the modifer:  `uv pip install --no-deps <whl path>` 

--
NB: Jumped through these steps to get NetworkTables working for a project.

virtuald, 2025-02-04:
We distribute wheels for macOS, is there a reason they don't work for you? Or are you just providing this because the instructions don't currently exist?

mikeohyan, 2025-02-04:
We thought you did! When trying to do a standard install (`uv add ...`), it didn't seem to be able to solve for MacOS. I think there was a wheel  dependency that wasn't solving correctly for the 2025.* versions. 

But also a nice to have, in case someone wanted to know where a wheel came from.

virtuald, 2025-02-04:
You should be able to do a `uv pip install pyntcore` to get ntcore support for macOS, there is a wheel published at https://pypi.org/project/pyntcore/#files

It is possible that our dependencies are weird and the uv project support might not be functioning, I haven't used uv very much.

We probably should document the build process better, maybe in build.md in the root directory.

auscompgeek, 2025-02-04:
My team uses uv for our robot code, and everything resolves correctly. I know because I tested it on both Linux and macOS 😄

TheTripleV, 2025-02-04:
I do understand that it's a bit confusing considering on pypi
- ntcore is unrelated
- networktables doesn't exist
- pynetworktables is abandonware but nowhere does it point to pyntcore
- pynetworktables2js is abandonware

mikeohyan, 2025-02-04:
> You should be able to do a `uv pip install pyntcore` to get ntcore support for macOS, there is a wheel published at https://pypi.org/project/pyntcore/#files
>
> It is possible that our dependencies are [...]

Just digging into `uv` for a bunch of new builds right now. It's the first manager that feels pretty good for python.

Happy to support, glad super responsive ppl are working on this. GL w/ your bots!