robotpy/mostrobotpy #165: [BUG]: hal notifier functions return non-zero status

MikeStitt, 2025-04-23:
### Problem description

The functions

```
hal.initializeNotifier
hal.setNotifierName
hal.updateNotifierAlarm
hal.waitForNotifierAlarm
```

all return non-zero status in python, while similar c++ code returns 0 status.

e.g.:

```
        self.notifier, status1 = initializeNotifier()

        status2 = setNotifierName(self.notifier, "TimedRobot")

        status3 = updateNotifierAlarm(self.notifier, startTime+100000)

        now_us, status4 = waitForNotifierAlarm(self.notifier)

```

### Operating System

MacOS

### Installed Python Packages

```text

```

### Reproducible example code

```text
from wpilib import IterativeRobotBase
import hal

from hal import initializeNotifier, setNotifierName, observeUserProgramStarting, updateNotifierAlarm, \
    waitForNotifierAlarm
from wpilib import RobotController

class ExpTimedRobotPy(IterativeRobotBase):

    def __init__(self, periodS: float = 0.020):  # todo are the units on period correct?
        super().__init__(periodS)

    def startCompetition(self) -> None:
        self.robotInit()

        if self.isSimulation():
            self._simulationInit()

        # Tell the DS that the robot is ready to be enabled
        print("********** Robot program startup complete **********")
        observeUserProgramStarting()

        startTime = RobotController.getFPGATime()

        status1 = 0
        status2 = 0
        status3 = 0
        status4 = 0

        self.notifier, status1 = initializeNotifier()

        status2 = setNotifierName(self.notifier, "TimedRobot")

        status3 = updateNotifierAlarm(self.notifier, startTime+100000)

        now_us, status4 = waitForNotifierAlarm(self.notifier)

        endTime = RobotController.getFPGATime()

        print(f"HAL_InitializeNotifier  : Notifier: {self.notifier} status=0x{status1&0xFFFFFFFF :08x}")
        print(f"HAL_SetNotifierName     : Notifier: {self.notifier} status=0x{status2&0xFFFFFFFF :08x}")
        print(f"HAL_UpdateNotifierAlarm : Notifier: {self.notifier} status=0x{status3&0xFFFFFFFF :08x}")
        print(f"HAL_WaitForNotifierAlarm: Notifier: {self.notifier} status=0x{status4&0xFFFFFFFF :08x}")
        print(f"GetFPGATime: startTime={startTime} now_us={now_us} endTime={endTime}")

    def endCompetition(self):
        hal.stopNotifier(self.notifier)
```

virtuald, 2025-04-23:
Your C++ code is explicitly initializing the status to zero, but I believe the wrappers do not. Thad says these functions do not explicitly set the status value unless there's an error, so that's probably the problem. Requires an update to robotpy-build, but I should be able to push that tonight.

(Thanks to Thad for pointing this out)

MikeStitt, 2025-04-23:
Some of the Java versions of these functions do not return a status. 

That might also be appropriate.

virtuald, 2025-04-24:
Please try installing https://github.com/robotpy/robotpy-build/pull/254 and see if that addresses your issue.

MikeStitt, 2025-04-27:
I'm might not be skilled enough to do the install.

I did this install:

```
git clone git@github.com:robotpy/robotpy-build.git
cd robotpy-build
git checkout zero-init-out
pip uninstall robotpy-build
python -m pip install -e .
cd ..
```

Then I get this:

```
command=cd mostrobotpy
command=python -m devtools develop
running develop
/Users/mikestitt/Documents/first/pythonExperiments/tryRobotPyBuildEditable/.venvTryRobotPyBuildEditable/lib/python3.12/site-packages/setuptools/command/develop.py:41: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  easy_install.initialize_options(self)
/Users/mikestitt/Documents/first/pythonExperiments/tryRobotPyBuildEditable/.venvTryRobotPyBuildEditable/lib/python3.12/site-packages/setuptools/_distutils/cmd.py:90: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
running egg_info
writing robotpy_wpiutil.egg-info/PKG-INFO
writing dependency_links to robotpy_wpiutil.egg-info/dependency_links.txt
writing entry points to robotpy_wpiutil.egg-info/entry_points.txt
writing requirements to robotpy_wpiutil.egg-info/requires.txt
writing top-level names to robotpy_wpiutil.egg-info/top_level.txt
writing manifest file 'robotpy_wpiutil.egg-info/SOURCES.txt'
running build_ext
running build_gen
running build_dl
WARNING: some items not in generation yaml for DataLog.yml
classes:
  StructLogEntry:
    methods:
      Update:
      HasLastValue:
      GetLastValue:
  StructArrayLogEntry:
    methods:
      Update:
      HasLastValue:
      GetLastValue:

WARNING: some items not in generation yaml for SendableRegistry.yml
classes:
[... 319 lines cut ...]
pyfrc                    2025.1.0
PyNaCl                   1.5.0
pynetconsole             2.0.4
pytest                   8.3.5
pytest-reraise           2.1.2
PyYAML                   6.0.2
requests                 2.32.3
robotpy                  2025.3.2.0
robotpy-build            2025.1.1.dev1+g330ea40 /Users/mikestitt/Documents/first/pythonExperiments/tryRobotPyBuildEditable/robotpy-build
robotpy-cli              2024.0.0
robotpy-installer        2025.1.1
robotpy-navx             2025.0.1.1
robotpy-wpilib-utilities 2025.0.0
setuptools               79.0.1
setuptools-scm           7.1.0
sphinxify                0.12
tomli                    2.2.1
tomli_w                  1.2.0
tomlkit                  0.13.2
toposort                 1.10
trpbe                    2025.0.1               /Users/mikestitt/Documents/first/pythonExperiments/tryRobotPyBuildEditable
typing_extensions        4.13.2
urllib3                  2.4.0
wheel                    0.45.1
```

auscompgeek, 2025-04-27:
That looks like you're missing the pybind11 submodule in robotpy-build. Try again after running `git submodule update --init`

MikeStitt, 2025-04-28:
Thanks @auscompgeek `` fixed my build problem.

Thanks @virtuald This fixed the status return problem. Now when trying the same robotpy simulation I get:

```
 robotpy sim
08:17:41:999 INFO    : faulthandler        : registered SIGUSR2 for PID 82672
08:17:42:001 INFO    : halsim_gui          : WPILib HAL Simulation 0.1.dev3186+g958c909.d20250428
HAL Extensions: Attempting to load: libhalsim_gui
Simulator GUI Initializing.
2025-04-28 08:17:42.045 Python[82672:11840806] WARNING: Secure coding is not enabled for restorable state! Enable secure coding by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState: and returning YES.
Simulator GUI Initialized!
HAL Extensions: Successfully loaded extension
08:17:42:083 WARNING : pyfrc.physics       : Cannot enable physics support, /Users/mikestitt/Documents/first/pythonExperiments/pythonRobot/physics.py not found
08:17:42:084 INFO    : wpilib              : RobotPy version 2025.3.2.0
08:17:42:084 INFO    : wpilib              : WPILib version 0.1.dev3186+g958c909.d20250428
08:17:42:084 INFO    : wpilib              : Running with simulated HAL.
08:17:42:085 INFO    : nt                  : Listening on NT3 port 1735, NT4 port 5810
Not loading CameraServerShared
********** Robot program startup complete **********
HAL_InitializeNotifier  : Notifier: 50331649 status=0x00000000
HAL_SetNotifierName     : Notifier: 50331649 status=0x00000000
HAL_UpdateNotifierAlarm : Notifier: 50331649 status=0x00000000
HAL_WaitForNotifierAlarm: Notifier: 50331649 status=0x00000000
GetFPGATime: startTime=171583 now_us=275552 endTime=275564
08:17:42:199 WARNING : your.robot          : Unexpected return from startCompetition() method.
```

closed 2025-04-30 by virtuald