robotpy/mostrobotpy #195: [BUG]: README develop directions don't work

truher, 2025-08-29:
### Problem description

I tried the "development environment" directions from the README:

```
./rdev.sh develop
```

This yielded the following:

```
+ /home/joel/ENV/robotpy/bin/python3 -m pip --disable-pip-version-check install -v -e . --no-build-isolation
Using pip 24.0 from /home/joel/ENV/robotpy/lib/python3.12/site-packages/pip (python 3.12)
Obtaining file:///home/joel/FRC/TRUHER/mostrobotpy/subprojects/robotpy-native-wpiutil
  Running command Checking if build backend supports build_editable
  Checking if build backend supports build_editable ... done
  Running command Preparing editable metadata (pyproject.toml)
  Traceback (most recent call last):
    File "/home/joel/ENV/robotpy/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 167, in prepare_metadata_for_build_editable
      hook = backend.prepare_metadata_for_build_editable
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  AttributeError: module 'hatchling.build' has no attribute 'prepare_metadata_for_build_editable'

```

### Operating System

Linux

### Installed Python Packages

```text

```

### Reproducible example code

```text

```

virtuald, 2025-08-29:
Duplicate of https://github.com/robotpy/mostrobotpy/issues/172

truher, 2025-08-29:
what i'm really trying to do is figure out why the python-wrapped apriltag library yields worse recall than calling the same library direction using apriltag_demo.  the input files are the same, the configs are the same, as far as i can tell.   so i was going to instrument the apriltag code being called by the python path, but that means figuring out both the apriltag library build, and the wpilib wrapper build, and the python build.   is there some sort of standalone way to do it?

virtuald, 2025-08-29:
If you install the two packages listed in #172 it should build. If it doesn't, that's a bug.

... I've only tested the develop bug on macOS and Linux

There currently isn't an easy way to build the apriltag library and integrate it with this. Fixing https://github.com/robotpy/hatch-robotpy/issues/2 would be the right way to make that possible.

truher, 2025-08-29:
Ugh.  I have failed to reproduce the problem with the Pupil wrapper, it works fine.  Same input file, same config.

If the WPILib apriltag thing came with a debug symbol file (or just a not-stripped debug build) I could poke at it with gdb.