robotpy/mostrobotpy #196: manylinux tags should be autodetected

virtuald, 2025-09-02:
Otherwise when we forget to update them then users get confused: https://www.chiefdelphi.com/t/robotpy-2027-error/505872/2

... for some reason I thought that hatch-meson set the tags correctly, but now I'm not so sure.

auscompgeek, 2025-09-02:
hatch-meson is able to detect for robotpy-native-wpiutil, not so sure about the others.

virtuald, 2025-09-04:
This project is pretty simple and is exactly what we need: https://github.com/hsorby/renamewheel

It uses auditwheel's internal API so it's broken with a current auditwheel installation. Also fails if the wheel is already properly named. Filed bugs to fix them, but if they don't respond this would be really easy to fork and put in one of our packages.

* https://github.com/hsorby/renamewheel/issues/1
* https://github.com/hsorby/renamewheel/issues/2